alexreinert/HB-RF-ETH #2: DCF-77 Timeserver ohne Funkmodul?

Triton-KS, 2020-09-19:
Hallo Alex,
So der Bausatz DCF-ET1 und DCF-Empfangsmodul DCF-2 habe ich zusammen gebaut.
Die LED signalisiert auch einen Empfang (ob der wirklich stimmt oder richtig ist kann ich irgendwie nicht testen)
Jetzt habe ich erstmal ohne Funkmodul HM-MOD-RPI-PCB versucht einen Windows 10 Rechner zu synchronisieren.
Funkmodul wird nicht erkannt, kan sein dass es einen Defekt hat.
Schläg aber mit der folgenden sinngemäßen Fehlermeldung "Aufgrund von Zeitüberschreitung konnte nich synchronisiert werden" fehl. Der Rechner und HB-RF-ETH hängen direkt an einem Switch und sind über DHCP im Netzwerk bekannt.
Liegt das am fehlenden Funkmodul? Kann ich es irgendwie testen ob der Timeserver richtig arbeitet?

alexreinert, 2020-09-20:
Der Timeserver arbeitet auch ohne Funkmodul. Allerdings antwortet er erst, wenn er entweder per RTC eine Uhrzeit beim Start erhalten hat oder wenn er mind. einmal eine Zeitsynchronisierung per eingestellter Zeitquelle durchgeführt hat.
Grade bei DCF kann das aber bei schlechtem Empfang durchaus etliche Minuten dauern, es braucht nur eine kurze Störung pro Minute vorkommen und man kann das komplette Frame der Minute nicht mehr verwenden. Selbst unter besten Empfangsbedingungen braucht es bis zu 2 Minuten, bis das erste vollständige Minutenframe empfangen wurde.

alexreinert, 2020-10-14:
Noch Fragen offen?

Triton-KS, 2020-10-14:
Hallo,
naja ich würde mich dennoch über eine Statusanzeige vom DCF77 / GPS Zeit freuen.
Auch wenn es nur eine Meldung "Ok / gestört" oder Anzeige der empfangenen Daten ist

closed 2020-10-14 by Triton-KS