alexreinert/HB-RF-ETH #3: Reconfigure Package

schmupu, 2020-10-06:
Leider hatte ich das Package `hb-rf-eth` installiert bevor die Hardware angeschlossen war. Jetzt habe ich das Paket mit `apt remove --purge hb-rf-eth` deinstalliert und anschließend mit `apt install hb-rf-eth` wieder installiert. Leider kommt das Konfigurationsmenü nicht mehr hoch um die richtige Hardware (IP) auszuwählen. Gibt es eine Möglichkeit die Konfiguration nochmals auszuführen?

alexreinert, 2020-10-06:
Sofern nur eine einzige HB-RF-ETH im Netz verfügbar ist, wird diese automatisch erkannt und genutzt. Das Auswahlmenü kommt nur, wenn mehr als eine HB-RF-ETH im Netzwerk verfügbar sind.

schmupu, 2020-10-06:
Hmm, das Menü kommt aber nicht mehr hoch beim `apt install hb-rf-eth`. Und das Modul ist im Netzwerk. Ich kann es mit http://192.168.20.168 aufrufen und habe auch ein Firmware Update auf 1.23 durchgeführt. Debmatic und HB-RF-ETH Modul sind im gleichen Netz. 
Unter Debmatic sehe ich das Modul nicht oder muss ich das noch irgendwie einbinden? Sorry, für meine Fragen!

alexreinert, 2020-10-06:
Wie viele HB-RF-ETH hast du im Netz? Wie geschrieben: Wenn nur eine, dann kommt kein Menü.

schmupu, 2020-10-06:
Genau ein Gerät! Okay, das erklärt warum kein Gerät angezeigt wird. Sehe ich in Debmatic, dass das Gerät  HB-RF-ETH erkannt wird?

alexreinert, 2020-10-06:
debmatic-info

schmupu, 2020-10-06:
Super, Danke es läuft! Danke nochmals für die Hilfe und für die tolle Platine und Debmatic. Ich bin begistert!

closed 2020-10-06 by schmupu