alexreinert/HB-RF-ETH #5: HB-RF-ETH verbindet sich nicht mit Debmatic

heinog, 2021-01-31:
Hallo, ich habe mir jetzt die HF-RF-ETH zugelegt und bin nach der Anleitung von technikkram.net gegangen.
Dh. Debmatic upgedatet auf VERSION=3.53.34.59.
Anschließend die Platine über POE ins Netzwerk gebracht und upgedatet. Danach Debmatic (Intel NUC) neu gestartet.
Die WEbui startet, es wird mir aber keine Firmware Version angezeigt.
Über die Oberfläche der Platine wird auch keine Verbindung angezeigt.
Was mache ich falsch?

alexreinert, 2021-01-31:
Wurde während der Installation des Pakets die IP Adresse der Platine angezeigt? Hat sich die Adresse nach der Installation geändert? Wird auf der WebUI der Platine das Funkmodul korrekt erkannt? Und zuletzt: Was sagt der Support vom Händler?

heinog, 2021-01-31:
Während des Updates von Depmatic hatte ich die Platine noch nicht angeschlossen.
Beim installieren von den addon sudo apt install hb-rf-eth wird mir immer angezeigt, das es schon vorhanden ist. Ob sich die IP geändert hat, weiss ich nicht. Auf der Webui der Platine wird mir RPI-RF-MOD angezeigt.
Den Support habe ich noch nicht erreicht.

alexreinert, 2021-01-31:
Der Platine eine feste IP verpassen, entweder direkt über die WebUI oder der fixer Zuweisung per DHCP im Router, dann auf der Konsole
`sudo dpkg-reconfigure hb-rf-eth` ausführen und danach debmatic neu starten.

P.S. debmatic und nicht depmatic

heinog, 2021-01-31:
Ok, werde ich morgen ausprobieren, habe jetzt erst mal den alten Zustand über USB wieder hergestellt. Sorry, war ein Tipfehler.
Vielen Dank erstmal, werde morgen berichten.
Gruß Heino Gerdes

heinog, 2021-02-01:
Hallo, ich habe es heute morgen ausprobiert. Die Verbindung ist jetzt da. Allerdinks ist die Grüne Powerled und die Rote Led jetzt auf Dauerlicht. Ist das Richtig?

alexreinert, 2021-02-01:
Welche LEDs genau? Die auf der HB-RF-ETH oder die auf dem RPI-RF-MOD?

heinog, 2021-02-01:
Die Grüne von der ETH und die Rote von der Mod

heinog, 2021-02-01:
Beim Hochlaufen ist sie erst auf Dauerlicht Grün und wechselt dann auf Dauerlicht Rot

alexreinert, 2021-02-01:
Grüne Power-LED ist normal.
Rote LED beim RPI-RF-MOD kann zwei Möglichkeiten haben: 
 - Es werden keine Keep-Alive von debmatic an die Platine gesendet, erkennt man, wenn man auf die WebUI der Platine geht und dort keine verbundene IP angezeigt wird
 - Der hssled Dienst von debmatic ist der Meinung, dass irgendwas innerhalb der CCU Prozesse nicht korrekt läuft. Da gibt es in Version 3.55.10 aber noch ein paar Optimierungen, damit das hoffentlich nicht mehr vorkommt.

heinog, 2021-02-01:
Ok, dann warte ich ab. Die Verbindung mit Debmatic wird mir angezeigt. Die RSSI Werte gemessen mit dem Homematic Manager sind auch besser als über USB.
Vielen Dank 
Heino Gerdes

heinog, 2021-02-11:
Hallo, nach dem Update war alles in Ordnung, die Led ist auf Blau.
Nun habe ich aber seit ein paar Tagen in der Debmatic eine Alarmmeldung, das sich der ETH nicht meldet. Ich kann ihn auch so quittieren. Die LED ist dann auch schon wieder blau. Woran kann das liegen.
Hier mal ein Auszug aus der Log Datei:
Feb  3 11:14:38 debmatic kernel: [180787.753301] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:38 debmatic kernel: [180788.389307] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:39 debmatic kernel: [180789.029298] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:40 debmatic kernel: [180789.669294] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:40 debmatic kernel: [180790.309278] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:41 debmatic kernel: [180790.949264] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:41 debmatic kernel: [180791.589248] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:42 debmatic kernel: [180792.229252] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:43 debmatic kernel: [180792.869239] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:43 debmatic kernel: [180793.513232] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:44 debmatic kernel: [180794.149223] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:45 debmatic kernel: [180794.789221] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:45 debmatic kernel: [180795.429194] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:46 debmatic kernel: [180796.069191] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:47 debmatic kernel: [180796.709184] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:47 debmatic kernel: [180797.349173] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:48 debmatic kernel: [180797.989157] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:48 debmatic kernel: [180798.629148] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:49 debmatic kernel: [180799.269138] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:50 debmatic kernel: [180799.909128] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:50 debmatic kernel: [180800.549114] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:51 debmatic kernel: [180801.189109] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:52 debmatic kernel: [180801.829096] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:52 debmatic kernel: [180802.469090] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:53 debmatic kernel: [180803.109086] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:54 debmatic kernel: [180803.749074] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:54 debmatic kernel: [180804.389058] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:55 debmatic kernel: [180805.029052] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:56 debmatic kernel: [180805.669041] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:56 debmatic kernel: [180806.309030] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:57 debmatic kernel: [180806.949025] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:57 debmatic kernel: [180807.589012] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:58 debmatic kernel: [180808.228997] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:59 debmatic kernel: [180808.868991] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:14:59 debmatic kernel: [180809.508978] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:15:00 debmatic kernel: [180810.148979] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:15:00 debmatic ReGaHss: SIGTERM: ReGaHss Halting
Feb  3 11:15:01 debmatic kernel: [180810.788969] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:15:01 debmatic kernel: [180811.432956] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:15:02 debmatic kernel: [180812.068962] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:15:03 debmatic kernel: [180812.708976] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:15:03 debmatic kernel: [180813.348966] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:15:04 debmatic kernel: [180813.988947] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:15:04 debmatic kernel: [180814.628914] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:15:05 debmatic kernel: [180815.268896] hb-rf-eth hb-rf-eth: Trying to reconnect to 192.168.178.37
Feb  3 11:15:06 debmatic kernel: [180815.845407] eq3loop: eq3loop_close_slave() mmd_hmip
Feb  3 11:15:06 debmatic kernel: [180815.851454] eq3loop: eq3loop_close_slave() mmd_bidcos
Feb  3 11:15:06 debmatic kernel: [180815.859019] eq3loop: eq3loop_close_master() mmd_bidcos
Feb  3 11:15:06 debmatic kernel: [180815.859024] eq3loop: eq3loop_close_master() mmd_bidcos destroy device
[... cut ...]
Feb  3 11:15:11 debmatic kernel: [180821.085860] hb-rf-eth hb-rf-eth: Trying to connect to 192.168.178.37
Feb  3 11:15:11 debmatic kernel: [180821.192811] hb-rf-eth hb-rf-eth: Successfully connected to 192.168.178.37
Feb  3 11:15:11 debmatic kernel: [180821.213766] raw-uart raw-uart: Reset radio module
Feb  3 11:15:14 debmatic kernel: [180824.321336] eq3loop: created slave mmd_hmip
Feb  3 11:15:14 debmatic kernel: [180824.322353] eq3loop: created slave mmd_bidcos
Feb  3 11:15:16 debmatic kernel: [180826.353967] eq3loop: eq3loop_open_slave() mmd_bidcos

closed 2021-04-19 by stale[bot]