alexreinert/HB-RF-ETH #11: Firmware update funktioniert nicht

FrankOverIP, 2021-03-14:
Hallo!
Nach der ersten Inbetriebnahme ist die HF-RF-ETH über Browser erreichbar, allerdings gelingt mir kein Update. Ich erhalte immer den Hinweis, dass das Update erfolgreich war und nach einem Neustart zur Verfügung steht. Tatsächlich wird aber nach dem Neustart weiter die alte Version gezeigt. Ich bin etwas ratlos...

Die HF-RF-ETH ist über USB mit Strom versorgt, der Netzwerkanschluss ist für PoE ausgerüstet, wird darüber aber (noch) nicht versorgt.

Hat vielleicht jemand eine Idee?

alexreinert, 2021-03-14:
Gescheites Netzteil mit mind. 1A oder PoE nutzen. Wenn das nicht funktioniert, Support vom Händler nutzen bzw. Quelle der Platine nennen.

FrankOverIP, 2021-03-14:
Netzteil war ein 1A Apple Ladegerät, ein 2,1A Apple hat nichts geändert. PoE ist im Moment future use, d.h. der aktuelle Switch kann das noch nicht. Die Platine kommt von smartkram.de.

alexreinert, 2021-03-14:
Dann Händler fragen, die haben schließlich Geld dafür bekommen.

closed 2021-05-20 by stale[bot]