alexreinert/HB-RF-ETH #12: HB-RF-ETH synology docker

handy80, 2021-03-27:
Hi
trying to install pivccu3 as a docker on my synology DS1813+ to use it in combination with the HB-RF-ETH board. 
I got issues to install the following as there is no apt in my synology. 
anyone knowledged/educated how i could solve that
sudo apt install wget ca-certificates
wget -q -O - https://www.pivccu.de/piVCCU/public.key | sudo apt-key add -
sudo sh -c 'echo "deb https://www.pivccu.de/piVCCU stable main" >/etc/apt/sources.list.d/pivccu.list'
sudo apt update
sudo apt install build-essential bison flex libssl-dev
sudo apt install pivccu-modules-dkms

kind regards

alexreinert, 2021-03-27:
This is not supported, piVCCU3 does not run inside Docker, piVCCU is supported on ARM platform only, Kernel >= 4.14 is required and Kernel Headers are needed.

handy80, 2021-03-27:
thanks Alex!

closed 2021-03-27 by handy80