alexreinert/HB-RF-ETH #14: Connect - bei Änderung der CCU IP Adresse?

thorsten-l, 2021-04-06:
Hallo Alex,

merkt sich die HB-RF-ETH von welcher IP Adresse sie ursprünglich mal angesprochen wurde? (Sicherheits-Feature)

Folgende Konfigs habe ich ausprobiert
- piVCCU3 (Pi4+Raspbian), funktioniert mit HB-RF-ETH (erster Test)
- Ubuntu 18.04 (amd64) + Raspberrymatic (Docker) - Keine Verbindung zur HB-RF-ETH
- Raspbian (armhf) + Raspberrymatic (Docker) - Keine Verbindung zur HB-RF-ETH

Alle piVCCU Module sind geladen und der avahi daemon hat auch auf allen drei Sytemen die IP Adresse des HB-RF-ETH
angezeigt und in /etc/defaults/hb_rf_eth eingetragen. /dev/raw-uart und /dev/eq3loop exisitieren auf allen drei Systemen.
Damit sich die Systeme nicht gegenseitig stören, war immer nur eines der drei eingeschaltet.

Da ich nun nicht mehr weiter weiß wollte ich zuerst Dich nach einem Tip fragen und danach Jens Maus ... ist ja "sein" Docker Image ;-)

LOG
```
ccu               | Starting watchdog...
ccu               | Identifying onboard hardware: oci, OK
ccu               | Initializing RTC Clock: no hardware found
ccu               | Running sysctl: OK
ccu               | Checking for Factory Reset: not required
ccu               | Checking for Backup Restore: not required
ccu               | Initializing System: ln: /dev/raw-uart: File exists
ccu               | ln: /dev/eq3loop: File exists
ccu               | OK
ccu               | Starting logging: OK
ccu               | Populating /dev using udev: done
ccu               | Init onboard LEDs: init, OK
ccu               | Starting irqbalance: OK
ccu               | Starting network: eth0: link up, fixed, firewall, inet up, 172.19.0.2, OK
ccu               | Identifying Homematic RF-Hardware: ...HmRF: none, HmIP: none, OK
ccu               | Updating Homematic RF-Hardware: no GPIO/USB connected RF-hardware found
```

alexreinert, 2021-04-06:
> merkt sich die HB-RF-ETH von welcher IP Adresse sie ursprünglich mal angesprochen wurde? (Sicherheits-Feature)

Jein. Innerhalb einer Verbindung werden nur die Pakete angenommen, welche zu der Verbindung passen. Allerdings ist es jederzeit möglich eine komplett neue Verbindung zu initiieren.

Zu RaspberryMatic kann ich keinen Support leisten, keine Ahnung was Jens da genau implementiert.

thorsten-l, 2021-04-06:
Das dachte ich mir schon. Danke für die schnelle Antwort.

closed 2021-04-06 by thorsten-l