alexreinert/HB-RF-ETH #18: LAN-Anbindung defekt nach 2 Tagen Internetstörung

AndreasBerz, 2021-05-11:
Gerätekonfiguration: HB-RF-ETH mit PoE-Modul und RPI-RF-MOD
HB-RF-ETH war auf DHCP konfiguriert.

Wir hatten 2 Tage Internetausfall wegen einer Störung im Telefonnetz.
Die LED auf dem RPI-RF-MOD hat schnelles blaues Blinken angezeigt.
Als ich einen Tag später wieder nachgeschaut habe war das Blinken weg, aber das HB-RF-ETH bindet sich nicht mehr ins LAN ein.
Stromversorgung über PoE scheint noch vorhanden, auch die Blink-Codes des HB-RF-ETH scheinen noch zu funktionieren.
Werks-Reset brachte keine Besserung. Alternativer Anschluss über None-PoE LAN mit USB-Versorgung auch nicht.

Das HB-RF-ETH scheint noch halbwegs zu funktionieren, sonst wären die Blink-Code beim Booten oder Werks-Reset nicht erkennbar. Aber das HB-RF-ETH geht nicht mehr ins LAN. Die Fritzbox zeigt kein passendes aktives IP-Gerät und auch der Netgear-Switch sagt, dass kein Gerät an seinem Port hängen würde.

Kann man davon ausgehen, dass sich der Ethernet-Controller des HB-RF-ETH verabschiedet hat?

alexreinert, 2021-05-12:
Wird der LAN8720A (IC U2) sehr warm? Dann dürfte dieser eine Macke haben.

AndreasBerz, 2021-05-13:
Der LAN8720A wird nicht außergewöhnlich warm. Aber natürlich können ICs auch Defekte haben, ohne warm zu werden.
Ich habe das HB-RF-ETH jetzt mal bei smartkram.de reklamiert. Es ist erst 0,5 Jahre alt.
Wenn es blöd läuft, ist das RPI-RF-MOD die Fehlerursache und nicht das HB-RF-ETH.
Da aber da der LAN-Controller auf dem HB-RF-ETH sitzt, gehe ich eher von einem Defekt des HB-RF-ETH aus.
Ein Versuch wäre noch das Neu-Flashen des HB-RF-ETH über den PC. Aber dazu braucht man das entspr. Knowhow und das Firmware-Image. Ist auch fraglich, ob das was bringt, wenn der Factory-Reset auch keine Besserung gebracht hat.

AndreasBerz, 2021-05-22:
Neues HB-RF-ETH bestellt und zusammengelötet, Funkmodul umgesteckt. Läuft!
Warum jedoch ein zweitägiger Internetausfall das HB-RF-ETH schädigen soll, will mir nicht in den Kopf.
Vermutlich gab es Spannungsspitzen, weil ich den Ethernetstecker (mit PoE) ein-, zweimal rausgezogen und reingesteckt habe.

peter-mul, 2021-07-21:
Im laufenden Betrieb hat sich auch meinen HB-RF-ETH 1.8 verabschiedet.

Es wird keinen Netzwerk-Link gezeigt. Nur der orangene LED leuchtet an der Netzwerkbuchse.
Die rote und grüne LED auf der Platine leuchten wie erwartet. Auch ein Reset läuft wie erwartet ab.
Der Netzwerkchip U2 wird nicht unerwartet warm.

Der Ersatz rpi4 darf jetzt mal wieder einspringen.

closed 2021-09-26 by stale[bot]