alexreinert/HB-RF-ETH #28: debmatic findet hb-rf-eth nicht

StefTank, 2022-01-17:
Hallo, 
hb-rf-eth ist im netz erreichbar. Firmware wurde auf 1.2.9 gebracht.
debmatic wurde in einer neuen VM (nicht LXC) installiert. hb-rf-eth wird hier mit der richtigen IP erkannt.
[image]
Mehrfache Neustarts und wechsel von Debian auf Ubuntu brachte keine Abhilfe.
Ein sudo dpkg-reconfigure hb-rf-eth hilft auch nicht.
[image]
Jetzt brauche ich Hilfe....

dlmcm87, 2022-03-04:
Hallo, das Problem habe ich jetzt auch - nach upgrade debmatic 3.61.7.89 im proxmox container.

Host / Proxmox connected zum HB-RF-ETH. Debmatic im Container hat keine Verbindung mehr ... Gibt schon Erkenntnisse?Workarounds?

handy80, 2022-03-21:
hier scheint das selbe Problem vorzuliegen. Hatte ein neues Gateway in meinem Netzwerk installiert. PiVCCU3 und RaspberryMatic ausprobiert, aber keine INstallation findet die HB-RF-ETH. Obwohl ich direkt auf die kleine WebOberfläche der HB-RF-ETH draufkomme

Schlonzmann, 2022-03-23:
Wenn mann eine vollwertige VM unter Proxmox erstellt funktioniert es.
Musste sudo apt install HB-RF-ETH aber nochmals mit sudo apt reinstall hb-rf-eth
Ausführen weil er es beim erstenmal auch nicht gefunden hatte

closed 2022-06-10 by stale[bot]