alexreinert/HB-RF-ETH #31: Fehlende Files für Installation via Platform.io

stefanfejfar, 2022-03-06:
Meine HB-RF-ETH0 bekommt keine Netzwerkadresse mehr, die grüne LED vom Board leuchtet aber. Update via WebUI ohne IP natürlich nicht möglich, Factory Reset hat auch keinen Erfolg gebracht.
Nun habe ich mich dran gemacht und versuche via VSCode und Platform.io die Platine neu zu installieren.

Leider scheint im Repository das "dist" Verzeichnis zu fehlen, sodass es zu folgender Fehlermeldung kommt und der Build bzw. auch Upload nicht durchlaufen kann.

`Warning! Could not find file "main.1e43358e.css.gz"`
`Warning! Could not find file "main.1e43358e.js.gz"`
`Warning! Could not find file "index.html.gz"`
`Warning! Could not find file "favicon.26242483.ico.gz"`

Ich beiß mir hier leider die Zähne aus. Hab diverse Libraries nun installiert (Perl, Python, Espressif32, Arduino IDE, CMake, C++, etc.) und mich am Errror Log abgearbeitet, um meinen Rechner kompilierfähig zu bekommen. Leider komme ich nicht weiter und scheint nun an diesen Files zu liegen.

@alexreinert - ich weiß es ist ein Hobbyprojekt und ich bin dir unendlich dankbar für deine Arbeit und die Entwicklung der Platine. Dürfte ich dich um Hilfe bitten bzw. eine Ergänzung der Readme, was man beim USB-flashen alles beachten muss? Der Hinweis auf platformio und das, was über Google zum Debugging der Platine auffindbar ist, ist leider sehr dürftig.

Besten Dank!

stefanfejfar, 2022-03-06:
`fatal: not a git repository (or any of the parent directories): .git
CMake Error in src/CMakeLists.txt:
  Cannot find source file:

    C:/Git-Repo/HB-RF-ETH/webui/src/main.1e43358e.css.gz

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx

CMake Error in src/CMakeLists.txt:
  Cannot find source file:

    C:/Git-Repo/HB-RF-ETH/webui/dist/main.1e43358e.js.gz

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx

CMake Error in src/CMakeLists.txt:
  Cannot find source file:

    C:/Git-Repo/HB-RF-ETH/webui/dist/index.html.gz

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx

CMake Error in src/CMakeLists.txt:
  Cannot find source file:

    C:/Git-Repo/HB-RF-ETH/webui/dist/favicon.26242483.ico.gz

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx

CMake Generate step failed.  Build files cannot be regenerated correctly.`

stefanfejfar, 2022-03-06:
Output des Serialmonitors:

> I (33109) UpdateCheck: Start checking for the latest available firmware.
> E (33109) esp-tls: couldn't get hostname for :www.debmatic.de:
> E (33109) esp-tls: Failed to open new connection
> E (33109) TRANS_SSL: Failed to open a new connection
> E (33119) HTTP_CLIENT: Connection failed, sock < 0
> I (33119) UpdateCheck: There is no newer firmware available.

virusmoere, 2022-03-19:
Hi,

du musst vorher die webui per npm "bauen" bzw. die dependencies installieren. Hierfür brauchst du node.js. Dann sind auch alle bisher fehlenden Dateien da.

vigeland, 2022-04-07:
Hast du einen Tip wie man "baut".  Welche Pakete braucht man genau ?

alexreinert, 2022-04-07:
Es muss node.js auf dem System installiert sein und platform.io, allerdings in einer älteren Version.
Ich bin im Moment dabei, dass auf die aktuelle Version zu heben, laut Changelog von esp-idf könnte das auch dein Problem mit dem Netzwerk lösen.

vigeland, 2022-04-07:
Danke. 

Hänge jetzt an: 

/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\hb-rf-eth\src\sysinfo.o:(.literal._Z18updateCPUUsageTaskPv+0x8): undefined reference 
to `uxTaskGetSystemState'
/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\hb-rf-eth\src\sysinfo.o: in function `updateCPUUsageTask(void*)':
\VisualStudioSource\HB-RF-ETH/src/sysinfo.cpp:47: undefined reference to `uxTaskGetSystemState'

ich habe Core 5.2.5 und Home 3.4.1

vigeland, 2022-04-07:
Ich habe mal uxTaskGetSystemState 0 zurückgeben lassen. 
Ich vermute die Funktion ist für das Web zur Lastanzeige.

Ich bekomme dann das.

I (26) boot: ESP-IDF 4.3.2 2nd stage bootloader
I (26) boot: compile time 17:27:30
I (26) boot: chip revision: 1
I (29) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (36) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : DIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (74) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  3 ota_0            OTA app          00 10 00010000 001d0000
I (96) boot:  4 ota_1            OTA app          00 11 001e0000 001d0000
I (103) boot:  5 spiffs           Unknown data     01 82 003b0000 00050000
I (111) boot: End of partition table
I (115) boot_comm: chip revision: 1, min. application chip revision: 0
I (123) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=4899ch (297372) map 
I (239) esp_image: segment 1: paddr=000589c4 vaddr=3ffb0000 size=02c70h ( 11376) load
I (243) esp_image: segment 2: paddr=0005b63c vaddr=40080000 size=049dch ( 18908) load
I (252) esp_image: segment 3: paddr=00060020 vaddr=400d0020 size=78480h (492672) map 
I (431) esp_image: segment 4: paddr=000d84a8 vaddr=400849dc size=09f3ch ( 40764) load
I (448) esp_image: segment 5: paddr=000e23ec vaddr=50000000 size=00010h (    16) load
I (456) boot: Loaded app from partition at offset 0x10000
I (456) boot: Disabling RNG early entropy source...
I (468) cpu_start: Pro cpu up.
I (468) cpu_start: Starting app cpu, entry point is 0x400823d4
I (0) cpu_start: App cpu up.
I (483) cpu_start: Pro cpu start user code
I (483) cpu_start: cpu freq: 160000000
I (483) cpu_start: Application information:
I (487) cpu_start: Project name:     HB-RF-ETH
I (492) cpu_start: App version:      1.2.9
I (497) cpu_start: Compile time:     Apr  7 2022 17:25:58
I (503) cpu_start: ELF file SHA256:  f4056cc35a3e119b...
I (509) cpu_start: ESP-IDF:          4.3.2
I (514) heap_init: Initializing. RAM available for dynamic allocation:
I (521) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (527) heap_init: At 3FFB5048 len 0002AFB8 (171 KiB): DRAM
I (534) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (540) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (546) heap_init: At 4008E918 len 000116E8 (69 KiB): IRAM
I (554) spi_flash: detected chip: generic
I (557) spi_flash: flash io: dio
I (593) system_api: Base MAC address is not set
I (593) system_api: read default base MAC address from EFUSE
I (603) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (603) gpio: GPIO[23]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (613) uart: queue free spaces: 20
E (823) emac_esp32: emac_esp32_init(353): reset timeout
E (823) esp_eth: esp_eth_driver_install(204): init mac failed
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400856e3
file: "src/ethernet.cpp" line 88
func: Ethernet::Ethernet(Settings*)
expression: esp_eth_driver_install(&_eth_config, &_eth_handle)
E (843) esp_eth: esp_eth_increase_reference(400): ethernet driver handle can't be null
E (843) esp_eth: esp_eth_update_input_path(314): ethernet driver handle can't be null
E (853) esp_eth: esp_eth_ioctl(352): ethernet driver handle can't be null
I (863) esp_eth.netif.glue: c0:12:43:3f:4b:03
I (873) esp_eth.netif.glue: ethernet attached to netif
E (873) esp_eth: esp_eth_start(263): ethernet driver handle can't be null
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x400856e3
file: "src/ethernet.cpp" line 94
func: void Ethernet::start()
expression: esp_eth_start(_eth_handle)
E (903) RTC: No RTC found.
W (18913) HB-RF-ETH: Radio module could not be detected.

vigeland, 2022-04-07:
I (9893) PushButtonHandler: Factory Reset mode timeout.
I (9893) gpio: GPIO[23]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0       
I (9893) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (9903) uart: queue free spaces: 20
I (10033) esp_eth.netif.glue: XX ( Die passende MAC lt. dhcpd )
I (10033) esp_eth.netif.glue: ethernet attached to netif
I (10043) Ethernet: Started
E (10043) RTC: No RTC found.
I (16743) HB-RF-ETH: Detected HM-MOD-RPI-PCB:
I (16743) HB-RF-ETH:   Serial: XXX
I (16743) HB-RF-ETH:   SGTIN: XXX
I (16753) HB-RF-ETH:   BidCos Radio MAC: XXX
I (16753) HB-RF-ETH:   HmIP Radio MAC: XXX
I (16763) HB-RF-ETH:   Firmware Version: 2.8.6 
I (46863) UpdateCheck: Start checking for the latest available firmware.
E (46863) esp-tls: couldn't get hostname for :www.debmatic.de:
E (46863) esp-tls: Failed to open new connection
E (46863) TRANS_SSL: Failed to open a new connection
E (46873) HTTP_CLIENT: Connection failed, sock < 0  
I (46873) UpdateCheck: There is no newer firmware available.

Das bekomme ich nach dem Flash der 1.2.9 von dir. Kein Netzwerk link.

Welche Version von esp-idf wäre die richtige ?

vigeland, 2022-04-07:
Oder leichter, wie ist dein Zeitplan für  "Ich bin im Moment dabei, dass auf die aktuelle Version zu heben,"

vigeland, 2022-04-22:
Hast du eine Idee, wann es eine übersetztbare Version auf Github gibt, passend zum  esp-idf 4.3.2

alexreinert, 2022-05-07:
@vigeland Grade eben. Oder anders ausgedrückt: It's done, when it's done and tested.

vigeland, 2022-05-09:
Danke, teste ich mal diese Woche.

vigeland, 2022-05-09:
Dicht dran, am Upload hängt er noch.

Uploading .pio/build/hb-rf-eth/firmware_1_3_0.bin
esptool.py v3.3
Serial port /dev/cu.usbserial-21130
Connecting.....
Traceback (most recent call last):
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: f4:cf:a2:91:66:54
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
  File "/Users/abu/.platformio/packages/tool-esptoolpy/esptool.py", line 5387, in <module>
    _main()
  File "/Users/abu/.platformio/packages/tool-esptoolpy/esptool.py", line 5380, in _main
    main()
  File "/Users/abu/.platformio/packages/tool-esptoolpy/esptool.py", line 4802, in main
    esp.flash_set_parameters(flash_size_bytes(args.flash_size))
  File "/Users/abu/.platformio/packages/tool-esptoolpy/esptool.py", line 1056, in flash_set_parameters
    self.check_command("set SPI params", ESP32ROM.ESP_SPI_SET_PARAMS,
  File "/Users/abu/.platformio/packages/tool-esptoolpy/esptool.py", line 495, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/Users/abu/.platformio/packages/tool-esptoolpy/esptool.py", line 468, in command
    p = self.read()
  File "/Users/abu/.platformio/packages/tool-esptoolpy/esptool.py", line 413, in read
    return next(self._slip_reader)
StopIteration
*** [upload] Error 1

alexreinert, 2022-05-09:
Ich kann nur sagen, dass es auf einem frisch installierten platform.io auf einer Windows 11 Maschine jetzt problemlos läuft.

vigeland, 2022-05-23:
Geht, auf meinen Rechner war die Baudrate aus der Konfiguration zu hoch, 115200 ging es.
Leider kann man damit aber die Ausfallplatine ( Kein Netzwerk ) nicht wieder beleben.
Kannst den Punkt schließen. Danke für die Hilfe

closed 2022-08-12 by stale[bot]