alexreinert/HB-RF-ETH #32: Ethernet not detected

virusmoere, 2022-03-18:
Hi,

I just finished the HB-RD-ETh with PoE and powered it on the first time. The device is geting power but there is no LAN connection. When connecting via USB and LAN (without PoE) running the serial monitor I get the following output:

`
(27) boot: ESP-IDF 4.3.2 2nd stage bootloader
I (27) boot: compile time 14:03:26
I (27) boot: chip revision: 1
I (29) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (74) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  3 ota_0            OTA app          00 10 00010000 001d0000
I (96) boot:  4 ota_1            OTA app          00 11 001e0000 001d0000
I (104) boot:  5 spiffs           Unknown data     01 82 003b0000 00050000
I (111) boot: End of partition table
I (116) boot: No factory image, trying OTA 0
I (121) boot_comm: chip revision: 1, min. application chip revision: 0
I (128) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=48ab4h (297652) map
I (244) esp_image: segment 1: paddr=00058adc vaddr=3ffb0000 size=02c7ch ( 11388) load
I (249) esp_image: segment 2: paddr=0005b760 vaddr=40080000 size=048b8h ( 18616) load
I (258) esp_image: segment 3: paddr=00060020 vaddr=400d0020 size=78508h (492808) map
I (437) esp_image: segment 4: paddr=000d8530 vaddr=400848b8 size=0a2e0h ( 41696) load
I (454) esp_image: segment 5: paddr=000e2818 vaddr=50000000 size=00010h (    16) load
I (462) boot: Loaded app from partition at offset 0x10000
I (512) boot: Set actual ota_seq=1 in otadata[0]
I (512) boot: Disabling RNG early entropy source...
I (523) cpu_start: Pro cpu up.
I (523) cpu_start: Starting app cpu, entry point is 0x400823d8
I (0) cpu_start: App cpu up.
I (537) cpu_start: Pro cpu start user code
I (538) cpu_start: cpu freq: 160000000
I (538) cpu_start: Application information:
I (542) cpu_start: Project name:     HB-RF-ETH
I (547) cpu_start: App version:      1.2.9
I (552) cpu_start: Compile time:     Mar 18 2022 14:07:56
I (558) cpu_start: ELF file SHA256:  7cc5ea8393174a25...
I (564) cpu_start: ESP-IDF:          4.3.2
I (569) heap_init: Initializing. RAM available for dynamic allocation:
I (576) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (582) heap_init: At 3FFB5070 len 0002AF90 (171 KiB): DRAM
I (588) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (595) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (601) heap_init: At 4008EB98 len 00011468 (69 KiB): IRAM
I (609) spi_flash: detected chip: generic
I (612) spi_flash: flash io: dio
I (646) system_api: Base MAC address is not set
I (646) system_api: read default base MAC address from EFUSE
I (646) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (656) gpio: GPIO[23]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (666) uart: queue free spaces: 20
E (866) emac_esp32: emac_esp32_init(353): reset timeout
E (866) esp_eth: esp_eth_driver_install(204): init mac failed
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400856e7
file: "src/ethernet.cpp" line 88
func: Ethernet::Ethernet(Settings*)
expression: esp_eth_driver_install(&_eth_config, &_eth_handle)
E (886) esp_eth: esp_eth_increase_reference(400): ethernet driver handle can't be null
E (886) esp_eth: esp_eth_update_input_path(314): ethernet driver handle can't be null
E (896) esp_eth: esp_eth_ioctl(352): ethernet driver handle can't be null
I (906) esp_eth.netif.glue: 04:13:43:3f:76:03
I (916) esp_eth.netif.glue: ethernet attached to netif
E (916) esp_eth: esp_eth_start(263): ethernet driver handle can't be null
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x400856e7
file: "src/ethernet.cpp" line 94
func: void Ethernet::start()
expression: esp_eth_start(_eth_handle)
E (946) RTC: No RTC found.
`

virusmoere, 2022-03-19:
Hier noch die Ausgabe der hier geposteten, bereits kompilierten Firmware:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6216
load:0x40078000,len:14488
load:0x40080400,len:3416
entry 0x40080638
I (317) cpu_start: Pro cpu up.
I (317) cpu_start: Application information:
I (317) cpu_start: Project name:     HB-RF-ETH
I (320) cpu_start: App version:      1.2.9
I (324) cpu_start: Compile time:     Feb  1 2021 22:00:32
I (330) cpu_start: ELF file SHA256:  e196b0f4ca5f5f1b...
I (336) cpu_start: ESP-IDF:          3.40200.210118
I (342) cpu_start: Starting app cpu, entry point is 0x4008244c
I (0) cpu_start: App cpu up.
I (352) heap_init: Initializing. RAM available for dynamic allocation:
I (359) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (365) heap_init: At 3FFB4388 len 0002BC78 (175 KiB): DRAM
I (372) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (378) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (384) heap_init: At 4008ED10 len 000112F0 (68 KiB): IRAM
I (390) cpu_start: Pro cpu start user code
I (430) spi_flash: detected chip: generic
I (430) spi_flash: flash io: dioI (455) system_api: Base MAC address is not set
I (455) system_api: read default base MAC address from EFUSE
I (455) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (465) gpio: GPIO[23]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (475) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (485) uart: queue free spaces: 20
E (605) lan8720: lan8720_init(393): wrong chip ID
E (605) esp_eth: esp_eth_driver_install(199): init phy failed
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40085407
file: "src/ethernet.cpp" line 88
func: Ethernet::Ethernet(Settings*)
expression: esp_eth_driver_install(&_eth_config, &_eth_handle)
E (625) esp_eth: esp_eth_increase_reference(385): ethernet driver handle can't be null
E (625) esp_eth: esp_eth_update_input_path(307): ethernet driver handle can't be null
E (635) esp_eth: esp_eth_ioctl(345): ethernet driver handle can't be null
I (645) esp_eth.netif.glue: 5c:34:0d:80:70:5f
I (645) esp_eth.netif.glue: ethernet attached to netif
E (655) esp_eth: esp_eth_start(257): ethernet driver handle can't be null
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x40085407
file: "src/ethernet.cpp" line 94
func: void Ethernet::start()
expression: esp_eth_start(_eth_handle)
E (685) RTC: No RTC found.
```

alexreinert, 2022-03-19:
> E (605) lan8720: lan8720_init(393): wrong chip ID
> E (605) esp_eth: esp_eth_driver_install(199): init phy failed
> ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40085407

Falsche oder defekte Hardware

virusmoere, 2022-03-19:
Danke für die Rückmeldung. Ein Fall für smartkram.de? Die Platine ist gerade neu.

alexreinert, 2022-03-19:
Es ist schwer, dass nur aufgrund eines Github Tickets zu beurteilen bei einer Platine, welche smartkram nur als Bausatz verkauft. Aber tendentiell ja.

virusmoere, 2022-03-19:
Ok, ich frage da mal nach. Wirklich viel machen kann ich dann ja sonst nicht.

agross, 2022-05-17:
Nach dem vierten Reboot einer heute zusammengebauten Platine habe ich dasselbe Verhalten. Am Port des Switches leuchtet das LED für PoE, aber das für Daten nicht. Und per DHCP holt er sich auch keine Adresse.

agross, 2022-05-23:
@virusmoere Hat sich smartkram bei dir gemeldet? Ich habe bisher keine Antwort bekommen.

virusmoere, 2022-07-30:
@agross ich habe kommentarlos eine neue Platine zugeschickt bekommen. Seitdem keine Probleme mehr.

closed 2022-10-08 by stale[bot]