alexreinert/HB-RF-ETH #35: Verbindungsabbruch

schroedolf, 2022-04-17:
Hallo, 

nach der Inbetriebnahme des Moduls läuft dieses für einen unregelmäßigen Zeitraum normal. 
Danach bricht die Verbindung mit dem Fehlercode 113 ab. 
Das Modul ist mit der POE erweiterung bestückt. 
Komisch ist auch, dass die Orange LED der RJ45-Buchse nicht leuchtet. 
Hier noch das Logging:

13:57:58.587 -> ̌⸮F/⸮⸮⸮F⸮J⸮ֆF⸮⸮⸮yz⸮)!F⸮FH1[0;32mI (2998) HB-RF-ETH: Detected RPI-RF-MOD:[0m
13:57:58.587 -> [0;32mI (2998) HB-RF-ETH:   Serial: 5A4993F2C0[0m
13:57:58.587 -> [0;32mI (2998) HB-RF-ETH:   SGTIN: 3014F711A0001F5A4993F2C0[0m
13:57:58.587 -> [0;32mI (3008) HB-RF-ETH:   BidCos Radio MAC: 0xFFF2C0[0m
13:57:58.587 -> [0;32mI (3008) HB-RF-ETH:   HmIP Radio MAC: 0xB4BCC6[0m
13:57:58.587 -> [0;32mI (3018) HB-RF-ETH:   Firmware Version: 4.4.12[0m
13:58:00.247 -> [0;32mI (4648) Ethernet: Link Up[0m
13:58:00.247 -> [0;32mI (4648) Ethernet: HW Addr f4:cf:a2:95:e0:2f[0m
13:58:04.180 -> [0;32mI (8608) esp_netif_handlers: eth ip: 192.168.1.23, mask: 255.255.255.0, gw: 192.168.1.1[0m
13:58:04.180 -> [0;32mI (8608) Ethernet: IPv4: 192.168.1.23[0m
13:58:28.469 -> [0;32mI (32878) SystemClock: Updated RTC to 2022-04-17 11:58:28 UTC[0m
13:58:28.703 -> [0;32mI (33118) UpdateCheck: Start checking for the latest available firmware.[0m
13:58:31.545 -> [0;32mI (35948) UpdateCheck: There is no newer firmware available.[0m
13:58:36.231 -> [0;32mI (40648) Ethernet: Link Down[0m
13:58:40.211 -> [0;32mI (44648) Ethernet: Link Up[0m
13:58:40.211 -> [0;32mI (44648) Ethernet: HW Addr f4:cf:a2:95:e0:2f[0m
13:58:42.242 -> [0;32mI (46648) Ethernet: Link Down[0m
13:58:50.239 -> [0;32mI (54648) Ethernet: Link Up[0m
13:58:50.239 -> [0;32mI (54648) Ethernet: HW Addr f4:cf:a2:95:e0:2f[0m
13:58:52.218 -> [0;32mI (56648) Ethernet: Link Down[0m
13:58:56.203 -> [0;32mI (60648) Ethernet: Link Up[0m
13:58:56.203 -> [0;32mI (60648) Ethernet: HW Addr f4:cf:a2:95:e0:2f[0m
13:59:01.700 -> [0;32mI (66108) esp_netif_handlers: eth ip: 192.168.1.23, mask: 255.255.255.0, gw: 192.168.1.1[0m
13:59:01.700 -> [0;32mI (66108) Ethernet: IPv4: 192.168.1.23[0m
14:17:08.192 -> [0;33mW (1152648) httpd_txrx: httpd_sock_err: error in recv : 113[0m
14:17:08.192 -> [0;32mI (1152648) Ethernet: Link Down[0m
14:17:11.134 -> [0;31mE (1155588) RawUartUdpListener: Connection timed out[0m
14:17:14.214 -> [0;32mI (1158648) Ethernet: Link Up[0m
14:17:14.214 -> [0;32mI (1158648) Ethernet: HW Addr f4:cf:a2:95:e0:2f[0m
14:17:16.209 -> [0;32mI (1160648) Ethernet: Link Down[0m
14:17:24.178 -> [0;32mI (1168648) Ethernet: Link Up[0m
14:17:24.225 -> [0;32mI (1168648) Ethernet: HW Addr f4:cf:a2:95:e0:2f[0m
14:17:26.222 -> [0;32mI (1170648) Ethernet: Link Down[0m
14:17:34.186 -> [0;32mI (1178648) Ethernet: Link Up[0m
14:17:34.186 -> [0;32mI (1178648) Ethernet: HW Addr f4:cf:a2:95:e0:2f[0m
14:17:41.637 -> [0;32mI (1186108) esp_netif_handlers: eth ip: 192.168.1.23, mask: 255.255.255.0, gw: 192.168.1.1[0m
14:17:41.685 -> [0;32mI (1186108) Ethernet: IPv4: 192.168.1.23[0m
14:18:06.204 -> [0;32mI (1210648) Ethernet: Link Down[0m
14:18:06.204 -> [0;33mW (1210648) httpd_txrx: httpd_sock_err: error in recv : 113[0m
14:18:06.204 -> [0;33mW (1210648) httpd_txrx: httpd_sock_err: error in recv : 113[0m
14:18:06.204 -> [0;33mW (1210648) httpd_txrx: httpd_sock_err: error in recv : 113[0m
14:18:06.204 -> [0;33mW (1210658) httpd_txrx: httpd_sock_err: error in recv : 113[0m
14:18:06.252 -> [0;33mW (1210668) httpd_txrx: httpd_sock_err: error in recv : 113[0m
14:18:09.431 -> [0;31mE (1213868) RawUartUdpListener: Connection timed out[0m
14:18:14.192 -> [0;32mI (1218648) Ethernet: Link Up[0m
14:18:14.192 -> [0;32mI (1218648) Ethernet: HW Addr f4:cf:a2:95:e0:2f[0m
14:18:16.190 -> [0;32mI (1220648) Ethernet: Link Down[0m
14:18:24.188 -> [0;32mI (1228648) Ethernet: Link Up[0m
14:18:24.188 -> [0;32mI (1228648) Ethernet: HW Addr f4:cf:a2:95:e0:2f[0m
14:18:26.179 -> [0;32mI (1230648) Ethernet: Link Down[0m
14:18:34.180 -> [0;32mI (1238648) Ethernet: Link Up[0m
14:18:34.180 -> [0;32mI (1238648) Ethernet: HW Addr f4:cf:a2:95:e0:2f[0m
14:18:36.223 -> [0;32mI (1240648) Ethernet: Link Down[0m
....

Danke für die Hilfe.

alexreinert, 2022-04-17:
Alle Lötstellen kontrolliert?

schroedolf, 2022-04-17:
Hallo und danke für die schnelle Antwort. 

Ja Lötstellen sehen für mich alle gut aus! 
Ich muss noch etwas ergänzen... 
Es scheint als würde die Orange LED an der RJ-45 Buchse nur leuchten, wenn die Energieversofgung über PoE erfolgt und die grüne LED nur leuchten wenn die Energieversorgung über USB erfolgt. 
(Eine Versorgung über das Funkmodul habe nicht getestet.)

Danke.

vigeland, 2022-04-22:
Das gleiche Problem hatte mein Board auch ( Das Up/down nur bei POE ), aber erst nach 2 Monaten.
Nachts ausgefallen. Effekt keine IP mehr. 
Flashen, nachlöten/messen , POE/USB Power nix zumachen. 

Neues Board, ging sofort.

@alexreinert : Hast du mal so ein Board in der Hand gehabt? Kommt ja an und an mal vor 
Fast immer POE glaube ich, wahrscheinlich nutzen 99 % POE.

schroedolf, 2022-05-01:
dann werde ich wohl auch reklamieren... ich hoffe das neue modul hält dann länger als 2 Monate...

closed 2022-05-01 by schroedolf