alexreinert/HB-RF-ETH #36: hb-rf-eth findet Funkmodul nicht (genauer: avahi-browse findet es nicht)

bsafh, 2022-05-27:
Moin,

mein Installation findet das vorhandene Funkmodul nicht ..

Umgebung:
Debian 11 in VM unter kvm (TrueNAS Scale)
HB-RF-ETH hat die feste IP 192.168.80.129
HB-RF-ETH ist von dem Debian aus ping-bar
Weboberfläche geht auch

Installation RaspBerryMatic als Add-On in HomeAssistant (aber soweit komme ich ja nicht) gemäß https://github.com/jens-maus/RaspberryMatic/wiki/Installation-HomeAssistant

pivccu-modules-dkms/stable,now 1.0.77 all  [installiert]

root@homeassistant:~# cat /etc/modules-load.d/eq3_char_loop.conf
eq3_char_loop

root@homeassistant:~# lsmod | grep eq
eq3_char_loop 

root@homeassistant:~# service pivccu-dkms status
● pivccu-dkms.service - piVCCU DKMS Modules
     Loaded: loaded (/lib/systemd/system/pivccu-dkms.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Fri 2022-05-27 21:11:32 CEST; 1min 11s ago
    Process: 60225 ExecStart=/var/lib/piVCCU/dkms/ensure_modules.sh (code=exited, status=0/SUCCESS)
   Main PID: 60225 (code=exited, status=0/SUCCESS)

Mai 27 21:11:32 homeassistant systemd[1]: Starting piVCCU DKMS Modules...
Mai 27 21:11:32 homeassistant systemd[1]: pivccu-dkms.service: Succeeded.
Mai 27 21:11:32 homeassistant systemd[1]: Finished piVCCU DKMS Modules.

root@homeassistant:~# ping 192.168.80.129
PING 192.168.80.129 (192.168.80.129) 56(84) bytes of data.
64 bytes from 192.168.80.129: icmp_seq=1 ttl=255 time=0.810 ms
64 bytes from 192.168.80.129: icmp_seq=2 ttl=255 time=0.816 ms
^C

**root@homeassistant:~# avahi-browse -p -r -t -k _raw-uart._udp
root@homeassistant:~#**

nix gefunden ....  :-(

avahi-browse liefert leider so rein gar nichts zurück :-(

[image]

[image]

Manuell die IP Adresse eintragen hilft leider auch nicht

root@homeassistant:~# cat /etc/default/hb_rf_eth
HB_RF_ETH_ADDRESS=192.168.80.129
root@homeassistant:~#

[image]

Jede Hilfe ist willkommen!

alexreinert, 2022-05-27:
Gleiches Subnet? Sonst funktioniert das nicht.
Ansonsten Doku von RaspberryMatic lesen und ggf. dort ein Ticket aufmachen. Da es dort nicht analog zur Referenzimplementierung (debmatic) eingebaut ist, supporte ich das nicht.

bsafh, 2022-05-27:
Gleiches Subnetz, ja

192.168.80.21 ist der Rechner mit der RaspberryMatic, 192.168.80.129 ist
der HB-RF-ETH

Ist allerdings ein Switch dazwischen. RaspberryMatic hängt direkt am
Backbone (Nortel 4548GT), von geht's über einen Netgear 24-port zum Modul).

bsafh, 2022-05-27:
Allerdings ist an der Stelle wo der Fehler auftritt, die RaspberryMatic
noch gar nicht im Spiel.

Nun gut.

bsafh, 2022-05-28:
Es gibt eine Teil-Auflösung:

Der Fehler hatte nichts mit RaspberryMatic zu tun.

1) im Backbone Switch (Avaya 4548GT, falls es wen interessiert) war `igmp snooping` an, aber dann blockt er alle anderen Multicast-Adressen - auch die für mDNS nötige `224.0.0.251`. 
Nachdem ich die mit einem obskuren CLI Befehl (`vlan igmp allow-<irgeendwas>-flood <vlan> <multicast-ip>` oder so) freigegeben hatte, erschien der RF-HB-ETH schon mal in der Liste von `mdns-scan` - aber nur auf dem Host der VMs (kvm/qemu auf TrueNAS Scale (Debian 11) - nicht in der virtuellen Maschine selbst :-(

Erst ein `/usr/sbin/ip link set dev macvtap0 allmulticast on` hat dann bewirkt, dass `mdns-scan` und `avahi-browse -a` auch in der VM (also dem HA-supervised) das Funkmodul anzeigen.

Dann geht auch die Installation des Debian-Pakets für `hb-rf-eth` dafür - auf dem HA-supervised-Debian in der VM.

Ob das nun auch noch im RaspberryMatic-HA-Addon ankommt - keine Ahnung. Das ist ja ein Docker-Container, wie ich das da rein leite - mal sehen. "Automatisch" geht das bestimmt nicht.

So sieht das dann aus:
```
root@homeassistant:/etc# cat /etc/default/hb_rf_eth
HB_RF_ETH_ADDRESS="192.168.80.129"
```

Vielleicht hilft's ja jemandem bei der Fehlersuche in ähnlichen Fällen.

Wenn das auch im Docker-Container so sein muss, kann man das ja dahin einblenden - irgendwie (also, zumindest mit docker-compose, aber das Docker-YAML dafür hat man ja nicht). Egal, denn: andere Baustelle (ist ja RaspberryMatic und hat nicht mehr mit dem Funkmodul zu tun)

Dies hat aber mit dem Modul zu tun:

Der HB-RF-ETH stirbt nach ein paar Minuten einfach (meistens) und dann auch per ping nicht mehr erreichbar (auch das Webinterface nicht mehr) - da hilft dann nur Neustart (Stecker ziehen). 

Das ist "ein wenig" unpraktisch...

bsafh, 2022-05-31:
Nun gibt's aber doch noch ein Problem, das garantiert nichts mit RaspberryMatic zu tun hat:

einmal das ETH- + das Funkmodul (im Charly-Gehäuse) an einen anderen Ort in der Wohnung gebracht, dabei natürlich stromlos gemacht.

Ergebnis: 
jetzt erkennt das HB-RF-ETH die auf dem Modul selbst aufgesteckte Funkplatine nicht mehr (obwohl ja die Stromversorgung erfolgreich über eben jenes Funkmodul läuft). (taucht in der Weboberfläche des ETH-Moduls nicht mehr auf, da steht nur "-" als Funkmodul). Auch nicht nach 10x stromlos machen.

Hardware-Fehler?

Muss eigentlich die grüne LED auf der Funkplatine (nicht dem ETH-Modul) leuchten oder nicht? Die blinkt nämlich nur 1x kurz auf sobald ich den Strom einstecke)?

Deutet das auf irgendwas bestimmtes hin?

Tuxtom007, 2022-06-16:
Ich hab ein ähnliches Problem.

Das Funkmodul wird bei mir überhaupt nicht mehr erkannt.

Bei mir war das Funkmodul lange Zeit über eine USB-Platine ans RaspberryMatic angeschlossen, welche als VM auf Proxmox lief - das funktioniert Monate lang einwandfrei.
Dann hab ich den Proxmox-Server auf neue Hardware umgezogen, alle VM darauf verschoben,  die USB-Platine wieder der VM zugeweisen und die wird nicht mehr erkannt - die grüne LED leuchtet dauernt.

Da ich das Teil aber eh an einer andere Stelle verlegen wollte, war die  HB-RF-ETH eine optimale Lösung. 
Platine aufgebaut soweit notwendig, Firmware-Update gemacht, die wird im Netz gefunden aber hier auch das Problem, das Funkmodule wird nicht erkannt.

Jetzt lese ich immer wieder, das irgentwas geflasht werden soll / muss - gibt das da irgentwo mal eine Anleitung was, wie und womit zu flashen ist ?

Ich bin kurz davor, das ganz rück zu bauen und wieder auf eine CCU3 zu wechseln

bsafh, 2022-07-13:
Ich nochmal ...

ich habe jetzt eine neue hb-rf-eth Platine. 

Neueste Firemware drauf (1.3.0).

Neu gestartet.

Ergebnis: Funkmodul wird nicht angezeigt.

LED auf dem ETH-Modul grün

LED auf Funkmodul blinkt 1x grün auf beim Anschließen des ELV-Netzteils an die Funkplatine und ist dann dunkel.

Gibt es irgendeine Chance, herauszufinden, welche der beiden Platinen denn nun spinnt?

closed 2022-09-24 by stale[bot]