alexreinert/HB-RF-ETH #39: Netzwerkeinstellungen inkl. DHCP scheinen nur im 192.168.178.1 Netzwerk zu funktionieren

mpietruschka, 2023-05-08:
Hej Alex,

ich habe den HB-RF-ETH an einem PoE Managed Switch laufen. Die Ports sind alle untagged. Das Gerät bekommt also nichts vom VLAN mit.

Im 192.168.178.1 Netzwerk kann ich das Gerät ohne Weiteres betreiben.

Nachdem ich es ins 192.167.181.1 Netz anschließe (selbes switch) holt sich das teil keine IP. Auch eine statisch eingestellte IP führt zu keiner Verbindung.

Drei Ansätze...

1. Reset mittels Tastenfolge
2. Downgrade
3. Firmware neu aufspielen, um evtl. Fehler in den gespeicherten Zuständen zu beheben oder auch ein Downgrade durchzuführen
4. Debug-Ausgaben prüfen

Ansatz 1: Erfolglos. Trotz mehrfachem Versuch.
Ansatz 2: Erfolglos. Wurde mit einem Fehler quittiert.

Ansatz 3:
Nachdem ich die von die von dir genutzte Espressif-Platform-Version gefunden hatte (4.1.0 ) scheiterte das aufspielen - an einem USB-Verbindungsproblem? 

Irgendwas mit '(...)wrong header(...)'.

Googeln sagt man soll den Boot Button gedrückt halten. Jetzt habe ich schon einiges an Zeit investiert und wollte einfach mal Fragen ;) Hast du den auf irgendwelchen Pins/Jumpern herausgeführt?

Apropos: Du Kannst die Platform in der platform.ini mit "platform = espressif32***@4.1.0***" festlegen. Sie wird dann automatisch installiert.

Ansatz 4: Kann ich USB- und POE-Stromversorgung gleichzeitig betreiben?

Kannst du mir vielleicht helfen?

Danke!

LG

alexreinert, 2023-05-08:
Da ich bei mir kein Netz 192.168.178.0 laufen habe, kann ich ziemlich sicher sagen, dass es auch mit anderen Netzbereichen läuft.
Aber neben dem eigenen VLAN braucht es Routing zwischen den VLANs und einen eigenen DHCP Server in dem VLAN. Ist das gegeben?

Es braucht den Boot-Button nicht, die Platinen enthält auch die Schaltung um per DTR und RTS den automatischen Upload durchzuführen und das funktioniert eigentlich extrem zuverlässig.

USB und PoE können gleichzeitig angeschlossen sein.

mpietruschka, 2023-05-12:
Vielen Dank für die schnelle Rückmeldung!

Für den Upload wurde per Default eine komische Geschwindigkeit genutzt - irgendwas mit 921600. Fälschlicherweise hatte ich angenommen **monitor_speed = 115200** würde dafür genutzt werden.

Nachdem ich in der platformio.ini **upload_speed = 115200** gesetzt hatte, funktionierte der Upload wie erwartet. Inkl. dem von Arduino bekannten DTR und RTS.

Allerdings konnte ich zum Netzwerk immer noch keine Verbindung aufbauen.

In der Seriellen Konsole konnte ich sehen, dass permanent zwischen "ethernet up" und "ethernet_down" gewechselt wurde. Auch am Switch ging das Lämpchen nicht an (PoE funktioniert immer). Ich schätze das Gerät kann sich mit dem Port nicht synchronisieren.

Sobald ich ein weiteres POE-Switch zwischen den Port und den HB-RF-ETH hänge, funktioniert alles. Inklusive der IP-Adresse.

Auch wenn ich den HB-RF-ETH zwei Ports weiter direkt anschließe, läuft alles.

Aus meiner Sicht scheint es ein elektrisches Problem zu sein.

Meist hängen in den großen Switches mehrere Ports an einem Chip. Womöglich verursacht eines der anderen Geräte oder der Chip selbst das Problem. Das hatte ich jetzt aber nicht mehr getestet.

Vielleicht finde ich morgen Zeit die angeschlossenen Geräte ordentlich herunterzufahren bevor ich sie vom Switch trenne und den HB-RF-ETH einzeln an dem problematischen Ports ausprobiere.

Hast du Erfahrungen mit solchen Phänomenen?

LG

mpietruschka, 2023-05-19:
Der Netzwerk-Port funktioniert nun gar nicht mehr. Hier scheint ein technischer Defekt vorzuliegen. Womöglich verträgt das Gerät das Trennen mit laufender PoE-Versorgung nicht so gut.

Ich habe jetzt ein Gerät ohne PoE bestellt. Mal sehen wie das sich verhält.

mpietruschka, 2023-06-05:
Nun hat mir Smartkram das bestellte Exemplar zugeschickt. Das funktioniert tadellos.

Das versorge ich mit einem PoE-Splitter. Damit liegt die Spannung des Netzwerkkabels nicht am Gerät selbst und ich kann mehr Einfluss auf das PoE-Netzteil nehmen.

Vielleicht braucht es im Design eine galvanische Trennung zwischen Ethernet-Port und Chip?

https://www.pulseelectronics.com/network-signal-transformers/ethernet-transformers/
https://www.mouser.de/c/?marcom=178569622&sort=pricing
https://www.mouser.de/ProductDetail/Pulse-Electronics/HX5008NL?qs=UCntq0WcAz6Edh1QmnthJQ%3D%3D

Grüße und Danke für die Rückmeldung!

mpietruschka, 2023-06-05:
Das Konzept war mir noch nicht bekannt. Aber, jetzt wo du es sagst, kann ich den HY931147C im Schema auch sehen.

Vielleicht sollte ich mal die Buchse austauschen *grübel*. Könnte das Sinn machen?

alexreinert, 2023-06-05:
Um welchen Switch handelt es sich denn?

mpietruschka, 2023-06-05:
HPE 1920-48G-PoE+ (370W) Switch JG928A

mpietruschka, 2023-06-05:
Gibt es bekannte Probleme mit PoE-Switches?

alexreinert, 2023-06-05:
Mir sind keine bekannt, war nur der Gedanke, ob ich ggf. Zugriff auf den gleichen Switch zum Testen habe. Ist allerdings nicht der Fall.

mpietruschka, 2023-06-05:
Das wäre auch tatsächlich etwas ungewöhnlich gewesen :D

Vor dem Produktiveinsatz hatte ich den Netzwerkanschluss gebraucht und deswegen den Stecken öfters - an die 20 Mal, denke ich - am HB-RF-ETH gezogen. Auf Smartkrams schrieb auch jemand etwas über die PoE Empfindlichkeit. Daher meine Mutmaßung.

https://technikkram.net/blog/2020/08/17/endlich-verfuegbar-neue-platine-hb-rf-eth-zusammenbau-und-einrichtung/#comment-164846

thuer-it, 2023-07-09:
Hallo zusammen,
Konntet ihr das Problem schon eingrenzen? Ich hab hier nämlich den selben Fehler. Lt Console kommt das LAN Port zwar hoch, am Switch (mit und ohne PoE) bleibt alles dunkel und die Platine bekommt keine IP Adresse. Liebe Grüße Andreas

mpietruschka, 2023-07-09:
Hallo thuer-it,

ich warte noch auf das Ersatzteil. Es sollte gegen Anfang August ankommen.

Tatsächlich hatte ich eine neue Platine ohne PoE bestellt und die funktioniert erstmal. Neubestellungen sind natürlich keine Lösung.

Sobald ich Neuigkeiten habe, werde ich mich hier melden.

LG

mpietruschka, 2023-07-25:
Um einen Defekt im PoE-Modul auszuschließen hatte ich es ausgelötet.

Nachdem ich das Modul nach langer Zeit erneut angeschlossen habe funktioniert es erstmal. Dann habe ich es erneut am Switch angeschlossen. Dann ging es wieder nicht mehr.

Die Netzwerkbuchse habe ich ausgetauscht. Aber leider hat das nichts gebessert.

thuer-it, 2023-07-25:
Ich habe Alex meine defekte Platine geschickt, damit er diese analysieren kann. Eventuell findet er ja den Fehler

andreas-bulling, 2023-12-18:
Gibt es Neuigkeiten diesbezüglich? Du scheinst nicht der einzige mit einem ählichen/gleichen Problem zu sein, wenn man die letzten Issues hier auf GitHub ansieht. Und auch bei mir läuft das Board seit ein paar Tagen plötzlich nicht mehr... @alexreinert Feedback bzw. Support wäre nett - die Issues sind schon länger unbearbeitet/-beantwortet...

closed 2025-01-21 by stale[bot]