alexreinert/HB-RF-ETH #41: HB-RF-ETH Verbindungsverlust nach einigen Tagen

Sargende1, 2023-10-16:
Hallo zusammen,

Ich bin langsam echt ratlos. Seit einigen Monaten kämpfe ich mit dem Problem das mein HB-RF-ETH Verbindungsprobleme mit meiner Proxmox VM bekommt. 2-3 Tage geht alles gut. Danach verliere ich die Verbindung und muss meine Raspberrymatic instanz neustarten. Dann geht das spiel wieder von Vorne los.

Ist das Problem bekannt oder kann ich hier etwas tun? Ich habe bereits versucht statt dem ELV-Netzteil das ganze per USB anzuschließen. Dort tritt das Problem ebenso auf.

Eine Lösung wäre jede nacht mein Raspberrymatic neuzustarten aber das ist nicht SInn der sache.

Ich bin für jeden Rat dankbar.

LG Jan

jfuchs1988, 2023-11-01:
Hallo, ich bekomme seit ein paar Wochen auch den Fehler im Rasberrymatic angezeigt (WatchDog: hb-rf-eth-connection - HB-RF-ETH: no connection), dass das Modul nicht erreichbar war.
Allerdings funktioniert das System weiterhin ohne Neustart.
LG
Johannes

andreas-bulling, 2023-12-18:
Hast du das Teil schonmal an ein externes USB Netzteil gehängt? Bei meinem Board (und evtl. bei mehreren anderen Usern) hat sich der PoE Port verabschiedet - zumindest die PoE Funktion. Mit externem USB Netzteil tut es bei mir...

closed 2025-01-21 by stale[bot]