alexreinert/HB-RF-ETH #46: HB-RF-ETH kann im Netzwerk nicht gefunden werden

recepucar, 2023-12-03:
hallo ich habe einen HB-RF-ETH mit POE. 
Der ist bei mir auf dem Dachboden installiert, Letzte Woche kam bei mir der Elektriker und musste mir eine neue Leitung für Licht ziehen. dafür war der HB-RF-ETH ohne Strom. Seit einer Woche bekomme ich keine Verbindung mehr zustande. Reset Klappt leider nicht.
gibt es eine bestimmte IP (Werks IP) nach der ich versuchen sie zu finden. oder eine Möglichkeit wie ich die FW neu aufspielen kann?

recepucar, 2023-12-06:
ich hab immer noch das Problem, ich habe schon versucht einen IP Scan zu machen ohne Erfolg,
Leider finde ich keine Anleitung die FW neu aufzusetzen. Wäre echt toll, es ist zwar ein neues unterwegs aber trotzdem ist es schade für 60€ ein neues bestellen zu müssen

andreas-bulling, 2023-12-18:
Seit ein paar Tagen bei mir das gleiche Problem. Von heute auf morgen scheint sich das Board keine IP Adresse mehr zu holen, Verbindung dadurch nicht mehr möglich.

vore, 2023-12-18:
> Seit ein paar Tagen bei mir das gleiche Problem. Von heute auf morgen scheint sich das Board keine IP Adresse mehr zu holen, Verbindung dadurch nicht mehr möglich.
> 
> Gerade habe ich das Board per externem USB Netzteil angeschlossen und, siehe da, tut alles. Die PoE Funktion bzw. der Port sind also offensichtlich kaputt

Moin! Habe auch mal eine Anfrage im Homematic-Forum gestellt und es scheint leider mehrere solcher Fälle zu geben. Bleibt anscheinend nur ein Neukauf :-(

andreas-bulling, 2023-12-18:
Nun ja, oder Tausch auf Kulanz - ist ja offensichtlich kein vereinzeltes Problem, sondern ein Problem einer ganzen Charge bzw. des Board-Designs (?) wie es scheint. Da würde ich schon erwarten, dass betroffene Kunden Ersatz bekommen.

recepucar, 2023-12-18:
Also ich hab das POE Modul ausgelötet, und probiert leider auch ohne erfolg. Ich hab jetzt ein neues Board bestellt ist auch angekommen aber sollte es noch einmal passieren nach einem jahr würde ich echt stinksauer sein.

andreas-bulling, 2023-12-18:
Ich bin schon jetzt alles andere als amused - da halten mehrere China-Boards von AliExpress in meiner Smart Home Installation länger...

alexreinert, 2023-12-18:
Ganz ehrlich, was erwartet ihr? Ich stelle das Zeug kostenlos komplett Open Source zur Verfügung. Wenn ihr ein Problem mit den Platinen von smartkram habt, dann klärt das mit denen, ich bin nicht der Produzent.
Und wenn euch die Teile dort zu teuer sind, dann lötet sie komplett selber oder lasst sie euch anderweitig bestücken.

andreas-bulling, 2023-12-18:
Wir erwarten eh nicht, dass du das für uns fixed bzw. wir von dir Ersatz bekommen. Welche meiner/unserer Aussagen hat bei dir diesen Eindruck erweckt?

Ich denke die Leute, die Issues zu diesem Problem (evtl. aber damit auch nicht zusammenhängenden/anderen Problemen) geöffnet haben, würden sich maximal über Rückmeldung freuen, ob hier ein Problem bei der Firmware vorliegen könnte. Oder ob sie sich direkt an smartkram wenden sollen.

vore, 2023-12-18:
Hi!
Sehe ich auch so. Ist keinerlei Vorwurf an Alex Reinert. Ich wollte eigentlich nur wissen, ob da was bekannt ist und ich etwas falsch gemacht haben könnte. Mir war aber auch nicht klar, dass Smartkram der Produzent ist. Das war eine hilfreiche Info.
Gruß

vigeland, 2024-06-20:
Nun hat es mit das zweite mal mit dem Fehler erwischt. 
Plötzlich geht nix mehr.
Kein Gewitter, kein Basteln, nix. Angeschlossen als POE. Am Switch liegt es nicht. Logging von anderen Switch und ohne POE.
Ursache die Karte bringt das Netzwerk nicht up. Kein Link. 

Kann es sein das POE das Ding irgendwann abschießt ?

I (4489) PushButtonHandler: Factory Reset mode started.
I (12489) PushButtonHandler: Factory Reset mode timeout.
I (12489) gpio: GPIO[23]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (12489) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (12499) uart: queue free spaces: 20
I (12629) esp_eth.netif.glue: f4:cf:a2:91:65:13
I (12629) esp_eth.netif.glue: ethernet attached to netif
I (12639) Ethernet: Started
I (12659) RTC: RX9130 RTC found and initialized.
I (12669) SystemClock: Updated time from RTC to 2024-06-20 06:14:34 UTC
I (14989) HB-RF-ETH: Detected RPI-RF-MOD:
I (14989) HB-RF-ETH: Serial:
I (14989) HB-RF-ETH: SGTIN:
I (14999) HB-RF-ETH: BidCos Radio
I (14999) HB-RF-ETH: HmIP Radio MAC:
I (15009) HB-RF-ETH: Firmware Version: 4.4.22
I (24639) Ethernet: Link Up
I (24639) Ethernet: HW Addr f4:
I (26639) Ethernet: Link Down
I (34639) Ethernet: Link Up
I (34639) Ethernet: HW Addr f4:
I (36639) Ethernet: Link Down
I (44639) Ethernet: Link Up
I (44639) Ethernet: HW Addr f4:
I (45109) UpdateCheck: Start checking for the latest available firmware.
E (45109) esp-tls: couldn't get hostname for :[www.debmatic.de](http://www.debmatic.de/):
E (45109) esp-tls: Failed to open new connection
E (45109) TRANS_SSL: Failed to open a new connection
E (45119) HTTP_CLIENT: Connection failed, sock < 0
I (45119) UpdateCheck: There is no newer firmware available.
I (46639) Ethernet: Link Down
I (54639) Ethernet: Link Up
I (54639) Ethernet: HW Addr f4:
I (56639) Ethernet: Link Down
I (64639) Ethernet: Link Up
I (64639) Ethernet: HW Addr f4:
I (66639) Ethernet: Link Down

closed 2025-01-31 by stale[bot]