alexreinert/HB-RF-ETH #50: Firmware Upgrade from 0.9.1 to 1.3.0

evendark, 2024-01-11:
Liebe Community, 

ich habe zwei neue Platinen und bei beiden kann ich kein Upgrade der Firmware vornehmen.
[image]

Der Upload der Firmware läuft problem los ebenso wie auch der Update Prozess.

[image]

Nach nach dem POE Reset über den Switch bzw. abstecken des Netzwerkkabels startet die Platine wieder mit der Original Firmware 0.9.1

Gleiches verhalten zeigt sich mit einem externen Netzteil und ohne POE oder auch mit anderen Firmware releases.

Gibt es hier ggf. eine Verbindung zu dem Issue: https://github.com/alexreinert/HB-RF-ETH/issues/39

Über Feedback würde ich mich sehr freuen.
Vielen Dank und herzliche Grüße

alexreinert, 2024-01-11:
Welche Datei wird hier genau hochgeladen?

evendark, 2024-01-11:
Hallo Alex,

diese:

https://github.com/alexreinert/HB-RF-ETH/releases/download/v1.3.0/firmware_1_3_0.bin

und diese zum testen:

https://github.com/alexreinert/HB-RF-ETH/releases/download/v1.0.1/firmware_1_0_1.bin

Vielen Dank für das schnelle Feedback.

alexreinert, 2024-01-11:
Ist ein Funkmodul aufgesteckt? Wenn ja, bitte während des Updates abstecken. Und gerne mal probieren, ob es funktioniert, wenn man zusätzlich ein Micro-USB-Netzteil anschließt.
Wenn es dann noch immer nicht geht, bitte an den Händler wenden.

evendark, 2024-01-11:
Hallo Alex,
vielen Dank, ich musste beides tun sowohl das Funkmodule abstecken als auch zusätzlich ein Micro-USB-Netzteil anschließen. 
Könntest du das ggf. bitte in dein https://github.com/alexreinert/HB-RF-ETH/blob/master/README.md mit aufnehmen? :)

closed 2024-01-11 by evendark